mkrabset/krabzcam #1: Upper case G-code

PetteriAimonen, 2020-11-08:
First of all, thanks for this great program! Finally there is a modern alternative to MakerCAM that is as easy to use.

I ran into a small issue with Smoothieboard: it ignores the lower case g-codes generated by KrabzCAM. Could KrabzCAM be made to output g-codes in upper case?

I could make a pull request for this, but it seems the repository does not have the source code of the project. Do you intend to release this as open source some time?

mkrabset, 2020-11-08:
Hi, thanks for your feedback!
I just added a checkbox "Generate uppercase GCode" above the "Save G-Code"-button. 
I don't wanna cause problems for other users, so I have left it disabled as default.

Best regards
Marius

PetteriAimonen, 2020-11-08:
Thanks, I tested it and works great!

If there is anything I can help with in this program, feel free to ping me :)

closed 2020-11-08 by PetteriAimonen