mkrabset/krabzcam #2: Feature request: drag knife corner actions

risacher, 2021-01-27:
I'd like to use KrabzCAM for a drag knife (I use a CB09/CB09U Graphtec Blade Holder in a CNC engraver).  

Since the cutting tip of a drag knife literally drags behind the center of motion, compensating actions need to be made to either close paths or square corners.  I believe these are generally called "corner actions".  

As a reference, Donek Tools offers a free gcode processor to add corner actions, but it's implemented as Excel macros.  (ugh) https://donektools.com/free-cnc-router-software/#corner-actions

mkrabset, 2021-01-27:
Thanks for the input!
I'll see what I can come up with, but no promises.
 
Currently I'm thinking something like a follow-path operation, but with parameters:
- dragOffset  (How far behind center is the tip of the knife)
- cutDepth  (Depth when cutting)
- swivelDepth (Depth when swiveling)
- feedRate
- plungeRate
- minSwivelAngle (Minimum turn angle for doing the 'corner action')

I guess for multiple path operations I will need to assume that the tool-angle is preserved during rapid moves.
Does this sound reasonable? (I don't have any experience with drag-knives)

mkrabset, 2021-01-29:
I've added a new operation type for drag knife. 
Please try it out. 
Feedback is appreciated.

risacher, 2021-02-01:
First, you are amazing, and I'm very impressed both with the drag_knife operation and with krabzcam in general.

I think the assumption that the tool-angle is preserved during rapid moves is a good assumption - it the only reasonable assumption.  There are machines called "tangent knives" that actually use the A axis to set the blade direction, but I ain't got one o' those.

Minor bug: Any "g1 z0" moves are specified as "g1 z0 fnan", which breaks the gcode.  I did a find-replace of "fnan" for "f100" in the gcode, and its running like a champ on my sainsmart genmitsu right now.

mkrabset, 2021-02-02:
Thanks for your nice words!
I made my own little 3d-printed dragknife to test it out. 
A bit tricky to set up, but quite interesting to see it in action.
 
The "g1 z0 fnan" bug should be fixed now.
Happy cutting!

risacher, 2021-02-02:
The actual result of my first cut with the dragknife operation came out well, but the path it cuts seems surprisingly jerky around curves, as if it's constantly over-correcting?  Is this expected?

[image]

mkrabset, 2021-02-02:
I would say this is expected.
There is quite a bit of math involved. :-)
Typically we start with a bezier curve, which is converted into a sequence of circular arc-segments and lines, given a tolerance limit for error.
(You can toggle showing these arcs by pressing 'c' while in the 2d viewport)
This interpolation often causes a tiny shift in direction when going from one segment to the next. This shift gets amplified by the tool-offset, causing these seemingly big adjustments in direction.

I'm not sure how best to achieve this, but I guess the more circular and smooth these bends are, the less adjustments required.
 
As you can see below, it is possible to join two circular segments with different radiuses without getting any adjustment at all.
[image]

It might help opening your SVG in inkscape and remove unnecessary path-nodes, or use the 'simplify path' feature.
(You can play with 'simplification threshold' in Edit / Preferences / Behaviour)

Hope this is helpful.

risacher, 2021-02-06:
I used this again for a different project and didn't see any of this jaggy-ness.  

Attached is the first cut I did with the KrabzCAM drag knife feature. (Logo of my son's hockey team)

[image]

closed 2021-02-06 by risacher