mkrabset/krabzcam #3: SVG parsing misses some "loose ends"

risacher, 2021-03-02:
I rendered some text in Inkscape with the Hershey Text (1-stroke) extension, and the font "EMS Readability". The resulting paths look fine in Inkscape, but when imported into Krabzcam, some of the letters are missing parts, such as "H", "h", "T" (et al.)  I speculate that this is a problem with the SVG parsing code.

Attached example file demonstrates.  (It has been renamed to a .txt file because GitHub doesn't like .svg)

[alpha-test-svg.txt](https://github.com/mkrabset/krabzcam/files/6070818/alpha-test-svg.txt)

mkrabset, 2021-03-02:
Thanks for your feedback!
Some of the path cleanup-code did too much cleanup :-)
The problem should be fixed now.

risacher, 2021-03-03:
Yep. Appears fix

closed 2021-03-03 by risacher