mkrabset/krabzcam #4: Reloading deleted path/svg

concretedog, 2021-03-04:
Hey, First of all, Krabzcam is excellent! Many thanks for it. I write for Hackspace magazine and I shall probably be mentioning it in an article in the next few months. One small issue I have found with it is that if you delete a selected path or svg you then can't reload the same file until you reload the page or close and reopen. It's the same on your github.io hosted version or on my local download. 

Massive kudos to you again. Easy to approach CAM is a massive part of making CNC stuff more accessible.

mkrabset, 2021-03-04:
Hi, thanks a lot for your feedback and your nice words!
The problem should be fixed now.
Seems to be browser-dependent, so nice if you can verify.

concretedog, 2021-03-15:
Hi sorry for massive delay! I realise I wasn't the best issue raiser there as I didn't state what I was using, it was happening in Chromium on ubuntu 20,04... anyway it's resolved, all working! Many thanks for a great tool, I'll ping you if/when I get it mentioned in the magazine as part of a small series I'm writing about a cheap CNC.

closed 2021-03-16 by mkrabset