mkrabset/krabzcam #5: Adding a speed parameter

korbav, 2021-03-29:
Hello, thank you for this wonderful.
I just tried it out and figured that no speed parameter was existing which results in a GCODE file with an unset M3 parameter.
Which is problematic since it means the mill will not turn at all and will break (at least in a CNC cut with a FOLLOW_PATH operation).

There might be a reason for the missing speed parameter in the UI?
If not I would recommend adding a field    Speed [...] rpm and using this parameter during the generation of the GCODE "M3 S10000" for example if the user sets the speed to 10000RPM

Thanks!

mkrabset, 2021-03-29:
Hi, thanks for your input!
I've added settings for spindle rpm now, with 10000 as default.

-Marius

korbav, 2021-03-29:
Wonderful! Thanks a lot for the reactivity!

xpy, 2021-03-30:
> Hi, thanks for your input!
> I've added settings for spindle rpm now, with 10000 as default.
> 
> -Marius

I am relatively new to these stuff so excuse me if I am very wrong, does that mean that you can use this to set different power for the laser as well?

mkrabset, 2021-03-30:
> > Hi, thanks for your input!
> > I've added settings for spindle rpm now, with 10000 as default.
> > -Marius
> 
> I am relatively new to these stuff so excuse me if I am very wrong, does that mean that you can use this to set different power for the laser as well?

Hi, I guess it depends on your controller firmware. 
I've added a setting for laser power as well, in the laser (follow path) operation, with 255 as default value.

xpy, 2021-03-30:
> Hi, I guess it depends on your controller firmware.
> I've added a setting for laser power as well, in the laser (follow path) operation, with 255 as default value.

It doesn't generate toolpaths now!
`Uncaught Wrong metrics {"p":[null,null],"c":[52.89047602398137,151.798258920271],"v":[null,null],"a":null}`

xpy, 2021-03-31:
> 
> 
> > > > > Hi, thanks for your input!
> > > > > I've added settings for spindle rpm now, with 10000 as default.
> > > > > -Marius
> > > > 
> > > > 
> > > > I am relatively new to these stuff so excuse me if I am very wrong, does that mean that you can use this to set different power for the laser as well?
> > > 
> > > 
> > > Hi, I guess it depends on your controller firmware.
> > > I've added a setting for laser power as well, in the laser (follow path) operation, with 255 as default value.
> > 
> > 
> > It doesn't generate toolpaths now!
> > `Uncaught Wrong metrics {"p":[null,null],"c":[52.89047602398137,151.798258920271],"v":[null,null],"a":null}`
> 
> Could you please provide a sample file?
> I'm not able to reproduce the error.

Sorry, I can also not reproduce it any more, might have been a cached file problem? It was just a test, I just put some circles around.

closed 2021-04-03 by mkrabset