mkrabset/krabzcam #6: What are default units?

andrewerrington, 2021-04-26:
Great software! But, when I open a DXF file with an object of known size, it appears at a different size in the main window.

For example, I have a square with rounded corners that I'd like to cut. The overall size of the square is 43 mm in QCAD but it appears in krabzcam as just under 26.5 mm to a side. I know that units are arbitrary, but it would be nice if one unit in DXF is the same as one unit in krabzcam.

Have I missed something? I've tried uploading the DXF here to illustrate the problem, but Github won't allow it.

Thank you.

mkrabset, 2021-04-26:
Thanks for your input!
I've done some changes that will hopefully solve the problem.
Notice that when loading dxfs (or svgs without unit information) the scale will depend on the selected unit in krabzcam.
Please verify.

-Marius

andrewerrington, 2021-04-26:
Perfect! My 43 mm square now covers 43 units on the screen. Thank you.

I am hoping for a very simple workflow:
Design in QCAD -> save DXF -> import DXF in krabzcam -> create toolpaths -> run gcode on machine

It's almost there, but when I select "Save G-Code" I only get one selected toolpath. Is that right?

For example, my workpiece looks like this:
[image]

So, I select the outline and choose "Profile Outside" to get the main shape, then I select all four holes together and choose "Profile Inside" to get the holes cut. Now, in "Toolpaths" I have two entries. When I click on "Save G-Code" I get a *.nc file containing only one toolpath (the toolpath which was selected). I am assuming that I have to make *.nc files for each toolpath and put them together myself. Is that correct?

This is great software, which I can now use for a real job. Thanks!

andrewerrington, 2021-04-26:
I just read the instructions. Yes, I can only export toolpaths one at a time. This is fine. :)

mkrabset, 2021-04-27:
Yes you can only save one toolpath at a time.
But in your case, if you can use the same tool, speed etc for both the cutout and the holes you can still do everything in one go.
Just select both the holes and the outer square, and use 'profile outside'.
When selected shapes are nested like this, the insides of the holes are considered 'outside'.

Like this:
[image]

You will not be able to control the cutting order though, so I guess tabs would be important unless you use vacuum or tape/glue for fixturing.

andrewerrington, 2021-04-27:
Ah! Very clever. I might use that.
Anyway, now that you handle the units from a DXF in the way I expected I will close this issue.
Thank you.

closed 2021-04-27 by andrewerrington