mkrabset/krabzcam #8: Editing parameters of generated toolpaths

geobruce, 2021-05-15:
Is it possible to edit a generated toolpath?
What I'm looking for is a feature to select a generated toolpath and edit the settings for that specific toolpath.

Thanks in advance
This is a very nice looking project and I hope I can replace JScut by this pretty project.

Keep up the great work.

mkrabset, 2021-05-16:
Hi, thanks for your feedback!

I'm afraid you cannot currently change setting on a toolpath after it has been generated. 
The only changes you can do to a toolpath after generation is optimization (optimize-button), and adding tabs on profiles (pressing 't')

Maybe you can explain a little bit about your workflow and about the scenario where this feature would be useful?

geobruce, 2021-05-18:
Hey thank you for your quick reply

I’m coming from JScut for 2D and for 3D I use Kiri:moto.

For example I was doing some tests with krabzcam I was pocketing out a pattern and after that I cut out the outline with another operation.
Example pictures https://photos.app.goo.gl/5A5biNzY8qSDxTHNA
After generating the pattern I ran it in a simulator and I noticed the gcode was not going how I wanted (not deep enough, wrong clockwise direction, wrong speeds…. Etc).
When I want to change this I have to do all my selections again and set all the speeds and feeds again.

It’s just so I can make quick changes.

mkrabset, 2021-05-19:
Hi again.
Nice designs on your photos!

I've just added support for editing toolpath-properties after generation. 
New 'edit'-button in the toolpath section. 

Hopefully I didn't break anything :-)
Please test it and let me know of any problems.

-Marius

geobruce, 2021-05-19:
Hey

Marius thanks for the addition. I really appreciate your effort. 
Ideally for me it would be more intuitive that the settings of the selected tool-path show up in the operation menu instead of a pop-up. That way it would be also possible to add other objects to the selection.
[image]

Is there a way I can contact you in private?

mkrabset, 2021-05-20:
Hi, 
I will keep the loose coupling between operations and toolpaths. 
References from toolpath to operation would cause too many problems :-).

But I've added a few new features that might help:

In the new 'edit toolpath properties'-dialog there is a new button for copying the changed properties back into the operations, either into an existing one (if name and type matches), or into a new one.
So if you have an existing toolpath and want to reuse the operation-properties for other toolpaths this can be used.

There is also a new 'apply'-button below the operation property list. When used, the property-values will be applied to all existing toolpaths having an operation with matching name and type.

In all cases where toolpaths are updated with new properties, tabs needs to be re-added (when relevant), and optimization have to be re-done.

Hope this is helpful :-)
-Marius

geobruce, 2021-05-23:
Looks good thanks!

closed 2021-07-05 by mkrabset

mgallelli, 2023-07-13:
Hi Marius,
How do you change the designation of the starting position in the toolpath? I have an scurve and the program seems to pick the bottom point each time. I dont see a hit when i search for "origin" in the documentation.

Cheers

mgallelli, 2023-07-13:
This is really a great tool you created btw!!

geobruce, 2023-07-14:
Maybe try to change clockwise and counterclockwise / conventional and climb cutting.