mkrabset/krabzcam #11: [feature request] V-carving + inlay

geobruce, 2021-05-23:
It would be wonderful to see this feature in an opensource online CAM package. For the moment I use the opensource software [F-engrave](https://www.scorchworks.com/Fengrave/fengrave.html).

This request exists out of 2 parts:

1. V-carving operation
Cut the desired shape with V-Bit
Parameters:
- V-bit angle
- V-bit max diameter (decides the max cutting depth)
- Step down depth
- If the V-bit is not able to clean all the needed stuff in one pass it can do multiple passes next to each other, you'll be left with some scallop height. 
- These lines of the scallop height can be cleaned up with a regular endmill.

2. Wooden inlay
Make a positive and negative shape that fit's in each other 1st step is to do a v-carve and next step you'll have to generate the part to inlay (negative part). 
An extra parameter you'll need here is the distance you have between the negative and positive part so that a saw blade can fit in.

JScut has already some sort of opensource v-carving. and F-engrave has all this functionality but their g-code is very poor it does a lot of things twice. And isn't optimized at all.

```
Krabzcam looks like a very promising package and I'd love to see this as my one stop project for all my 2d, 2.5d and carving and inlay work. 

I'd love to support this project anyway I can. Wheter it is by sending some Belgian beers to you, send you some money, helping where I can.

Over a few weeks I'll be teaching a CNC lesson in a FabLab where lot's others will be following remote and I'm planning to demo this project also.
```

mkrabset, 2021-05-23:
Hi, you read my mind. I was just watching a few youtube-videos about v-carving and wooden inlays this morning. :-)
It would make a great addition, so I will give it a try. 
Guess I need a new voronoi-based algorithm for the variable-depth part.
This will probably take some time..

geobruce, 2021-05-24:
> Guess I need a new voronoi-based algorithm for the variable-depth part.
At the bottom of this page are a few links with some more explanation and approaches: https://www.scorchworks.com/Fengrave/fengrave.html#changelog

mkrabset, 2021-05-24:
Read a few papers on the subject.. Pretty complex stuff!
But I found a nice library that looks promising: https://www.npmjs.com/package/flo-mat

So crossing my fingers and hoping it has everything I need... 🤞
[image]

geobruce, 2021-05-25:
Wow the schreenshot looks promising already!

[image]<https://user-images.githubusercontent.com/885864/119396368-0b629c80-bcd5-11eb-98d7-7b32f03fce9a.png>

mkrabset, 2021-05-30:
Hi, I've added a new operation 'VPocket'  😀
My first test here: https://www.flickr.com/gp/25948865@N03/Rw11z4   

Please try it out and tell me how it works.

geobruce, 2021-05-30:
Very nice your example looks very good.

From this point it can't be to far away to generate for wooden inlays

geobruce, 2021-05-30:
works like a charm!
Only you still need something to do the cleanup after the max z-depth was reached. In f-engrave you have the choice to do the cleanup with either the v-bit or a regular endmill.

keep up the great work! 

[image]

mkrabset, 2021-05-30:
In the top left corner of your screenshot it says 'Clearance: 2.3094'
You can use a regular pocket operation for the cleanup, and use this number as 'Roughing clearance'.

I will do the V-Inlay operation next...shouldn't be that much work. 
I just need to slow down a bit. Has been a few very late nights :-)

mkrabset, 2021-06-01:
Hi again. I've added the V-Inlay operation.

The maxDepth property has been renamed to flatDepth, and an extra property startDepth is added for V-Inlay.

For VPocket, flatDepth is the depth of the pockets
For VInlay, flatDepth is the sawblade-gap, while startDepth is how deep the plug goes into the pocket before the artwork-planes align.
I believe this is the same terminology used in the VCarve-software.

The v-inlay operation also has switches for automatic/manual horizontal flip and creation of a rectangular boundary.

I added an "Add Cleanup" button for generating endmill cleanup-toolpaths for both VPocket and VInlay. 
These generated operations can be edited (tool-diameter etc) like the other.

Please test it out and let me know if you notice any issues.

geobruce, 2021-06-01:
This looks like everything I've been looking for. I still need to test it on a physical machine. One question tough how does the boundary for the inlay work? (automatic / manual) I need to increase it in size it seems.
 
[image]

mkrabset, 2021-06-01:
Hi, when using automatic boundary, the distance between the box and the artwork will be 4 times the clearance-value. 
But the box is treated the same way as the other paths, so it gets its own clearance. So in between, the distance is 2 times the clearance.
It might be a bit tight, I don't know. 
It's also quite easy to do it manually...Just hit '4' to get a square, then scale and move it into place. Just remember to only include the boundary when generating the inlay.
...Or edit the cleanup toolpath and choose a smaller tool-diameter perhaps..

I'll probably add some improvements to the gcode-generation tomorrow...It currently travels a lot in the z-direction. 

Have fun, but keep an eye on the machine...there might be some glitches... :-)

mkrabset, 2021-06-02:
I've added the improved gcode-generation and also added a new parameter 'Boundary clearance' for v-inlay. This is only used when choosing automatic boundary box to set the box-size.
I've also removed the boundary-box paths from the v-inlay toolpaths. No need to walk these paths with the v-bit I guess.

geobruce, 2021-06-19:
I found a bug
[project.json.gz](https://github.com/mkrabset/krabzcam/files/6680790/project.json.gz)
[files video and exported enviroment.zip](https://github.com/mkrabset/krabzcam/files/6680792/files.video.and.exported.enviroment.zip)

[image]
The generated g-code doesn't go to the safe height when needed.

[image]
And when you optimize the gcode and export it again the code hasn't changed.

mkrabset, 2021-06-20:
Hi, yes that's a bug. Thank you, nice catch!
Wish there was a working version of CAMotics for Fedora 34 too :-(
Should be fixed now. Please verify.

geobruce, 2021-06-20:
in the simulation it seems to be fixed I only checked the v-carve and not the inlay part.

Thank you very much!!

geobruce, 2021-06-25:
I tested the v-carve in real life and it works like a charm!

Thanks for your effort!

I'd like to reach out to you but I can't find your email address. Please send me an email at brucehelsen (at) hotmail (dot) com

Thanks in advance!

closed 2021-08-22 by mkrabset

geobruce, 2022-11-08:
I get some unexpected behavior with a simple circle I would expect the v-bit to just make an inside circle with a certain offset from the original circle. I don't get why there are all this lines pointing outwards.
[image]
[image]
[test file 1.zip](https://github.com/mkrabset/krabzcam/files/9964974/test.file.1.zip)
[image]

extra: What does the parameter max corner angle change?
how to I set the spacing between the v-carve and the inlay for a saw to fit in between.

Thanks in advance!

mkrabset, 2022-11-18:
@geobruce I've just made some improvements on v-bit operations, hopefully solving your issues with the spikes inside the circular path. 
I've also removed the 'maxCornerAngle'-parameter and replaced with a better approach.
Please let me know of any issues.

geobruce, 2022-11-18:
Thanks I'll test this asap, another thing I noticed there were G1 commands generated that cut right through the stock material. These lines should have been g0 at safe height.

Can't give an example right now but also will do this when I can.

Rhank!s!

mkrabset, 2022-11-18:
The problem you describe sounds like a bug that was fixed a week ago. 
But if you are still able to reproduce it with the latest version then please let me know.

geobruce, 2022-11-18:
I had this bug the 9th of november

mkrabset, 2022-11-19:
The bugfix I mentioned was released the 10th.