mkrabset/krabzcam #9: [feature request] Rest machining

geobruce, 2021-05-23:
It would be nice to have a feature where you could do a pocket or any operation. With a rough large diameter bit first and only cleanup the left overs with a smaller diameter bit. This is useful for large operations with only a little text or details that must be kept. You can see an example of this functionality in [Easel](https://easel.inventables.com/users/sign_in) it is not opensource and you do need a free account. But I prefer to support open source software.

In your preview in krabzCam it's easy to see what material will be left over as the bit can't go there. Maybe it's not too much work for you to add this useful feature that will save many people much time.

mkrabset, 2021-05-23:
Hi, yes this would be a nice time-saver.
I will do some thinking. Probably a lot of complexity here..

mkrabset, 2021-06-05:
I've added rest-machinging functionality to the 'add cleanup'-button. This button can now be used for both v-bit toolpaths and profiles/pockets.

Please check it out and let me know of any issues.
[image]

geobruce, 2021-06-06:
This looks fantastic! Thank you very much!!

closed 2021-07-05 by mkrabset