mkrabset/krabzcam #12: [feature request] Automatic dogbone

geobruce, 2021-06-09:
Kiri:moto has a feature which I find pretty pleasant to have.
Where it checks whether if an end-mill can reach into a corner or not. If this is not possible it automatically places a dogbone in the corner if this option is enabled.
This is nice for two scenarios:

1. If you forgot to add dog-bones or extra clearance
2. If you take a random design from the internet, designed for laser cutting instead of cnc routing.

mkrabset, 2021-06-11:
Hi, sounds useful!
I'll have a look at this next week.

mkrabset, 2021-06-14:
Hi, I've added a new option 'add dogbones' for inside and outside profiles.
Please check it out and let me know of any issues.

mkrabset, 2021-08-22:
Hi Bruce, have you tested this yet? 
Maybe this issue can be closed now?

geobruce, 2021-08-23:
Hey 

I made a file to test this feature.
On the inside I get some unexpected results:
[image]

Outside works like expected.
[image]

Thanks for your effort! This will come in handy for sure!

mkrabset, 2022-03-06:
I've added a fix for the inside profile problem.

closed 2022-04-13 by mkrabset

geobruce, 2022-10-04:
I'm trying to use the dog-bone feature and I discovered an issue.

outside profile: (no dog-bones are generated) (project attached)
[profile outside.json.gz](https://github.com/mkrabset/krabzcam/files/9709869/profile.outside.json.gz)
[image]

profile outside inconsistent behavior: [outside inconsistent.json.gz](https://github.com/mkrabset/krabzcam/files/9709926/outside.inconsistent.json.gz)
[image]

Profile inside [dia = 10mm] (no dog-bones are generated) [prof-inside-10mm.json.gz](https://github.com/mkrabset/krabzcam/files/9709927/prof-inside-10mm.json.gz)
[image]

 Profile inside [dia = 1mm] (dog-bones are generated) with the default diameter it works)
[image]

[image]
[dogbone test.zip](https://github.com/mkrabset/krabzcam/files/9709876/dogbone.test.zip)

Can you please have a look at this?
Do you might know a work around or a quick fix?
I'm taking a day off soon to have some time with the CNC machine I'm planning on making this: 
[image]
I included a part of the design file I plan on cutting out. So you can test this. I'm planning on using a 8mm and 3.175mm router bit.
[extensive test file v0.1a.zip](https://github.com/mkrabset/krabzcam/files/9709954/extensive.test.file.v0.1a.zip)

Thanks in advance and keep up the good work!
Please let me know if you get notifications for closed issues. (I'm not sure if this is the right way the re open an issue)