mkrabset/krabzcam #15: [Feature request] Other Pocket paterns

geobruce, 2021-08-10:
Last weekend I played around with a pen holder on my CNC. and I tried using krabzCam for generating g-code. For drawing = engraving lines it's perfect. But for filling (=pocketing, I guess) the drawing it works kind of.  
[image]
[image]

It would be cool that the pocket feature would have other strategies as well like:
Line:
[image]
Trianagle:
[image]
Grid: = cross hatch

This might go two ways:
1. extended clearing strategies for pocketing
2. extended clearing strategies for pocketing + new specific operations for plotters
With a few nice features like cross hatching, hatching, ... this can go quite far: 
[image]

Please let me know what you think.

Kind regards
Bruce

mkrabset, 2021-08-22:
Hi, yes the pocketing toolpaths aren't optimal for plotter use.
I guess it would be possible to fill shapes using the 'RELIEF_FOLLOW'-operation, but this would be cumbersome and would also involve converting the svg to an image.

I'm thinking about something like:
* Adding a pattern attribute to the pocket operation, with 'DEFAULT' and 'LINE' as selectable values (dropdown).
* When selected pattern is 'LINE', show an additional 'angle' attribute.

...so in order to achieve the 'triangle'-pattern (above), you create 3 identical 'LINE'-toolpaths with angles 0, 60 and 120 deg.

It may also make sense to have an option for 'uni-directional', to always cut in the same direction instead of back and forth.

In addition to plotter usage, I think this could also be useful for surfacing operations.

What do you think?

geobruce, 2021-08-23:
> I'm thinking about something like:
>
>     * Adding a pattern attribute to the pocket operation, with 'DEFAULT' and 'LINE' as selectable values (dropdown).
>
>     * When selected pattern is 'LINE', show [...]

This sounds very interesting! I'll can do some tests if the feature is there ;)

> I guess it would be possible to fill shapes using the 'RELIEF_FOLLOW'-operation, but this would be cumbersome and would also involve converting the svg to an image.
> 

I've never tried this feature before what kind of image does it expect?

mkrabset, 2021-08-23:
The RELIEF_FOLLOW-operation uses a grayscale-image as height-map and moves the tool over the 'terrain'.
...like [this](https://www.flickr.com/photos/25948865@N03/49711856193/), creating [this](https://www.flickr.com/photos/25948865@N03/49712388746/) result.

With a black and white image and some gcode editing I guess you could achieve back-and-forth strokes with a pen...but as I said..it would be a bit cumbersome.

I will try to create LINE support for pocket-operation as described...but it will take some time. I'll let you know.

geobruce, 2021-08-23:
Thanks in advance !!

...like ***@***.***/49711856193/>, creating ***@***.***/49712388746/> result.

mkrabset, 2021-08-30:
Hi again. 
I've added support for 'line-pockets' in the latest version. (Hopefully without breaking anything)
Please try it out and let me know of any issues.

-Marius

geobruce, 2021-08-31:
I wasn't able to test this on the machine but in the browser it's looking good! Thanks for that.
Is there already a functionality to set the lines at a certain angle, because I couldn't find that.

Thanks for your effort keeping on improving the features!

mkrabset, 2021-08-31:
Hi, when you select pattern='Line', a few additional parameters should become visible..
* LineAngle
* LineDirection

[image]

Use of 'optimize' also seems to be useful.

-Marius

geobruce, 2021-09-01:
strange I didn't see it the first time maybe my version wasn't refreshed.

Thanks for adding this, it looks good I'll test this when I've time.

mkrabset, 2021-09-01:
Did some testing myself as well 🙂
Drew this little guy on my plotter.
Had to do some gcode editing though, since my plotter is using m5/m3 for pen up/down.

[image]

closed 2022-03-06 by mkrabset