mkrabset/krabzcam #16: Safety Height Issue

CycloRetro, 2021-08-22:
Hi
I'm a previous MakerCam user and I'v been testing KrabCam today and noticed a few issues or oversights
1. Safety Height
Its best treat the safety hieght or z axis the same as the x and y axis as 0. So when the x and y and z are at 0,0,0 the cutting bit is at your start point and above the piece as set by the user. So when the cutting bit reaches the target depth it will raise back to 0 move safely over your peice to the next shape to be cut and plunge to begin the second shape.
2. Size Issue
When a .svg file is imported into KrabCam I've notice a decrease in the graphic's dimensions, it imports at about 33% smaller than the original artwork. Obviuosly this is a problem especially if you're an engraver like myself who needs to engrave on very small objects and fit into exact spaces.
3. Offline Version
Also it would be great to have an offline version that stands alone installed on one's computer to future avoid what just happen with MakerCam or internet problems. I'd be more than happy to buy a downloadable version of KrabCam

Hope this helps
Chris

mkrabset, 2021-08-22:
Hi, thanks for your input!

1. If I understand correctly, you would want z=0 to be at safety height, instead of at the z-level where the tool touches the workpiece.
I'm afraid I have to disagree. For me it makes more sense to zero the machine where the tool touches the material (using a touch sensor or the paper-method or whatever). A sudden change in this behaviour would probably also cause accidents for existing users. 
So for now I will leave this as-is. 
Maybe I will reconsider later if I get the same issue raised by other users. As of now I haven't.

2. Sizing in SVG is a bit involved. Please provide a sample file and info about what sizes you expect, and I will have a look.

3. If you want a local copy, you can just open it and save it locally as 'complete webpage'. 
It has no server-side backend, so your local copy should contain everything you need. 

best regards
Marius

CycloRetro, 2021-08-25:
Hi Marius

the z height after doing a cut should always return to the z height set by
the operator which is usually just above the material. Currently your
KrabCam doesn't include code to do this, it currently moves to the next
shape to cut without raising the z axis. As you can imagine this causes the
cutting bit to smash into the material and break the cutting bit.

In regards to the size issue see attached my original svg file and the
gcode created in KrabCam

Kind regards

Chris Howard

P: 0407 841 088
E: ***@***.*** ***@***.***>
W: www.cycloretro.com
F: www.facebook.com/chrishowardcycloretro

CycloRetro, 2021-08-29:
> Still no attachment.

Do you have another email address

Chris

mkrabset, 2021-08-29:
Hi, it's not a good idea to write personal email-addresses here, since this conversation is publicly visible at github.

I suspect the problem is that you reply via email. 
Try sending attachement using the github web-interface instead. 
That should work fine as long as you follow the instructions at the link I sent you.

You will then also be able to preview your reply before you send it.

-Marius

closed 2022-03-06 by mkrabset