mkrabset/krabzcam #18: M3 vs M4 for laser

risacher, 2021-09-30:
Modern versions of GRBL have a "laser" mode, which makes a distinction between M3 and M4 for laser operations.  (See
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode) 

M3 turns the laser on in constant-power mode. 
M4 is "dynamic power mode" and "will automatically adjust laser power based on the current speed relative to the programmed rate. It essentially ensures the amount of laser energy along a cut is consistent even though the machine may be stopped or actively accelerating."

This could be handled in a post-processing step, but it might also be convenient if there was a way to do it right in KrabzCAM.

risacher, 2021-09-30:
Not sure the best UI for this.  Maybe separate operations for laser(cut) vs laser(engrave)?

mkrabset, 2021-09-30:
Hi, thanks for your input!
How about adding a new parameter '**Laser ON command**' for operation 'FOLLOW_PATH_LASER'...with a dropdown for choosing between **M3** and **M4**.
Would that solve this issue?
-Marius

risacher, 2021-10-12:
From my perspective, absolutely.

closed 2022-02-08 by risacher