mkrabset/krabzcam #20: Interior paths not generated

Hilldamn, 2021-11-08:
When I import my SVG and select profile outside for cutting it only generates the paths for the outer squares and not the inner shapes. Not sure if this is a problem with the SVG file or if I didn't specify a parameter. Here is a zip file of the SVG and a screenshot of issue.
[map.svg.zip](https://github.com/mkrabset/krabzcam/files/7495488/map.svg.zip)
[image]

mkrabset, 2021-11-08:
Hi, thanks for your input!
When creating profiles or pockets, all involved paths needs to be closed. 
All interior shapes in your drawing are open paths, so that is the reason.
I guess you need to close the open paths and edit them so the follow their perimeters.
(hope this makes sense :-) )

Best regards
Marius

Hilldamn, 2021-11-08:
Yup, that was the issue. Thank you so much!

closed 2021-11-08 by Hilldamn