mkrabset/krabzcam #21: [Feature request] Plasma paterns

geobruce, 2021-11-23:
Hello

I was thinking about something. One day I'd love to hook up a plasma cutter to my cnc machine.
Therefore it would be interesting to have a little extra in the toolpaths. a start point that you can choose on the in or outside of your object. something like this: https://torchmate.com/cad-cam-7-8-9/FAQs-Tool-Paths

This could be as simple as a little straight line that is placed at every vector that is customizable for the in and outside.

Do you consider this as something usefull? 

Thanks in advance

davi-domo, 2023-04-23:
Hi,
The idea is original but it represents a lot of work for a very specific use
Who knows you may have given Marius an idea

mkrabset, 2024-12-09:
Not sure if this is any help for plastma cutting, but I've added some support for creating external entry-points for profiles, using hotkey 'shift-e'.

[image]

geobruce, 2024-12-10:
Nice!
It seems that the location of this entery point is fixed and can not be chosen like when you are choosing a location for a tab.
It would be nice to have it like this.
1. Shift-e
2. Choose start point on vector (just like placing a tab)
3. control entry line

Thank a lot!

mkrabset, 2024-12-10:
@geobruce You can set the internal entrypoint first using "e" (instead of shift-e)

See https://mkrabset.github.io/krabzcam/manual/manual.html#_changing_entrypoints

geobruce, 2024-12-10:
thanks a lot works great!

closed 2024-12-10 by geobruce