mkrabset/krabzcam #22: License

Thanson16, 2022-01-19:
Hello,

Firstly, awesome project. My friend and I are very impressed with the ease of use and are hoping to use it in the near future.

I was going to make a few customizations for our use case, but noticed two things:

1) There isn't a license for the project (just for the dependencies from what I can see in `2.3b087336.chunk.js.LICENSE.txt`)
2) The JavaScript files appear to be minified

So, I am wondering if you could associate a license (perhaps in a LICENSE.txt or something similar), so that people like me know whether they are allowed to modify and/or redistribute the software, and also whether it would be possible to have the original, un-minified source files (assuming the license allowed modifications and such).

Thank you for your time.

vannoo67, 2023-11-23:
I 3rd that request, though looking at the dates of the previous posts, I'm guessing I'm on a fools errand.
Is this project actually Open Source?

mkrabset, 2023-11-23:
Sorry, I should have responded to this a long time ago.
The software is free of use and can be downloaded and run offline, but as of now the source code is closed.

best regards
Marius

vannoo67, 2023-11-23:
Thanks for the response. 
Are you planning on opening the source at some point in the future? I would very much like to extend it and possibly integrate it with something like ESP3D (WebUI used by FluidNC)

mkrabset, 2023-11-23:
I might opensource it sometime in the future, but no promises.
-Marius

vannoo67, 2023-11-23:
Fair enough. If you do, please comment here.
BTW: Fantastic work on the product, it really is a beautiful thing.

Thanson16, 2023-11-23:
Thank you for the reply and the great software, Marius!