mkrabset/krabzcam #23: [Feature request] comments in gcode

risacher, 2022-02-08:
I saved many gcode files for a project recently, and I had trouble remembering what starting heights I chose for the operations.   It might be nice to add comments into the gcode for the various operations, date/time and/or documenting the names of the SVG files used. 

```
; generated at 2022-02-08T23:08:11.049Z
; source files: dragon.svg logo.svg
; operation laser FOLLOW_PATH_LASER 200 mm/min 255 power 
g0 x208.90916549 y9.79353371
g1 z-1.8 f400
g3 i-0.15 j0 x208.62928714 y9.86857778 f800
g1 z-2 f400
g3 i-3.0803191 j-5.63091501 x206.90343384 y10.51149669 f800
...
```

mkrabset, 2022-03-06:
Thank you for the suggestion! 
I agree, such comments would be nice.
The problem is that different controllers use different syntax for comments. 
Some use braces, some use leading semicolons etc..
And I want to keep the UI simple...

risacher, 2022-06-12:
Gcode is standardized as ISO 6983-1:2009.  I wonder what the official comment format is in the standard, and whether that format would be consistently supported. Alas, access to ISO standards is expensive and I can't find a free source.  

I did find a preview online that included this text: 

> If there is any group of characters that is not to be processed in accordance with this part of ISO 6983, this group shall be within parenthesis characters (control out – control in). Any such group shall not contain either “:” or “%” characters. This group may be processed for display purposes, e.g. as instructions to an operator.

Can anyone identify a controller that *doesn't* support `(comments)` like this? (What are called "parentheses" in American English and "brackets" or "round brackets" in British English?)

risacher, 2022-08-07:
You are amazing!

closed 2022-08-07 by risacher