mkrabset/krabzcam #25: [Feature request] Calculate cut time

kirkrudolph, 2022-03-06:
One feature I've found extremely useful in other CAM software is the ability to add the cut/travel distance and divide by the respective feed rates to calculate how long the cut will take. Thanks for considering this feature. Awesome work!

mkrabset, 2022-04-13:
Thanks for your suggestion!
This would be nice, but is not trivial.
KrabzCAM does not communicate with the CNC hardware, so it cannot query for information about speed pr axis for rapid move, acceleration etc.
I think it might make more sense to have this functionality in the "gcode sender" software.

kirkrudolph, 2022-04-15:
Sure, that makes sense. Thanks for considering it. If anyone asks for a similar feature in the future, I found a [g-code simulator](https://nraynaud.github.io/webgcode/) that's giving me ballpark estimates and is pretty simple to use.

Again, great work!

closed 2022-04-15 by kirkrudolph