mkrabset/krabzcam #26: lift between separate engraved paths?

wcray, 2022-03-06:
Beautiful work!

Would it be possible to return the cutter to the safety height while traveling between unconnected SVG components in engraving mode?  Right now, travel between components is at whatever the current cutting depth is, as is the lead-in from home to the initial cut point.

(Or is there a reason why this is a desirable behavior?  I can't think of a reason, but sometimes I'm dense!)

If I get the chance I'll dive under the hood and see if I can tweak this myself - small price to pay for the great work you've done on this so far!

mkrabset, 2022-03-06:
Hi, returning to safety height between cuts is the intended behaviour. I'm not sure why that doesn't work for you. 
Maybe you can provide an example?

wcray, 2022-03-07:
And... It looks like I'm an idiot.

Your software is working fine, this is a non-issue.

It turns out that my problem is that the gcode previewer that I've been using on the Mac for ages (Pleasant3D), apparently doesn't correctly represent g0 moves.  I don't know how I've missed this for the past 7+ years!

My apologies, and again, kudos for the great work.  I develop and maintain some esoteric biological data-visualization software, and I'm painfully aware of the amount of effort you've invested and are continuing to invest.  Color me impressed!

Will Ray

closed 2022-03-07 by wcray