mkrabset/krabzcam #27: [Feature request] multiline select generation gcode

davi-domo, 2022-04-12:
Hello,
a multiline selection when generating the gcode in order to select all the lines.
Congratulations for his work

mkrabset, 2022-04-13:
If I understand your request correctly, this is already supported.
When having multiple toolpaths, you are able to select multiple ones AFTER pressing the "Save-G code" button...

In the "Save G-code for selected toolpaths" dialogue:
[image]

You can also change the ordering by using the up/down buttons.

davi-domo, 2022-04-13:
> Links:

Hello,
as a picture is worth words, here is a picture of my request
once again congratulations for your project I use it all the time
as soon as I have a free moment I even thought of doing the translation 
for French speakers. (with my amateur bases in development)
this tool is still little known but very appreciated by what discovers 
it, simple, intuitive, fast, are the feedback that I have been given

Cordially

mkrabset, 2022-04-13:
Hi again. I cannot see any picture in your message. 
Please use the github web UI to respond. I think that works better with attachments.

davi-domo, 2022-04-13:
OK sorry
[image]

mkrabset, 2022-04-13:
Thank you!
 I've added the checkbox now, in the table-header instead of at the dialogue title...
Please try it out and let me know of any problems.

davi-domo, 2022-04-13:
I just did a test with a big project and no problem
thank you for being quick
good day

closed 2022-05-29 by mkrabset