mkrabset/krabzcam #28: [Feature request] Select paths by color

risacher, 2022-06-12:
I usually generate SVG files as input to Krabzcam in Inkscape.  I think it would be useful to be able to select paths by color in Krabzcam so that I can indicate in Inkscape that different parts are intended for the same operation.  For example, in Inkscape, I might put all the *dark* laser engraves in blue, and all the *light* engraves in red.  Or similar for different CNC depths, whatever.

risacher, 2022-08-15:
Awesome.  Thanks for this.  Closing.

risacher, 2022-09-19:
Found a use case where this is not working properly.  It seems to select closed paths separately from open paths..?  See linked example and try to use Shift-A to select the text (which is all one color) [cat tag](https://risacher.org/moriko-35mm.svg). At least for me, this will select the 'o', '0', and the dot on the 'i' separately from the other letters.

davi-domo, 2022-09-19:
hello
have you tried with a smaller cutter diameter?
this has already happened to me if the cutter is larger than the surface to be removed

mkrabset, 2022-09-29:
I've added a fix for the closed vs. open path problem. 
Please test it, and let me know of any problems.

risacher, 2022-10-02:
As it turns out, I had to re-make the part where I found this bug.  The fix worked great.  Thanks and closing.

risacher, 2022-10-02:
@davi-domo When I found the bug, I was using it for laser engraving, so the cutter size is not relevant.

closed 2022-10-02 by risacher