mkrabset/krabzcam #29: [Feature request] - User documentation

davi-domo, 2022-06-28:
Hello,
I use KRABZCAM all the time in my YouTube videos
more and more subscribers find it really interesting and I understand them
I wanted to know if there was a detailed documentation of all the commands
Cordially

dadoprom, 2022-06-28:
could you please share your YT channel link, I would like to have a look :)

davi-domo, 2022-06-28:
Hello my YT Channel link https://youtube.com/channel/UCnXQ3Z28tpQ3FVL6E3TE6cw

mkrabset, 2022-06-28:
Hi, I've been watching (and like'ing) your videos for a while now.  
Nice work, and nice to see KrabzCAM in action in your projects !

I'm afraid I haven't paid much attention to documentation yet.
There is a blue "HELP" link on the application, displaying some quite old basic docs, 
but apart from that there is only the hotkeys displayed in the background of the 2d view.

It's on my todo-list though.. :-)

best regards
Marius

davi-domo, 2022-06-28:
hello, 

I understand, we're all running out of time, me being the first
I'm going to start a video tutorial because I think KrabzCAM should be known to makers, just like "candle" where the beta version is promising
I'll come back to you as soon as it's done

mkrabset, 2022-06-29:
Cool!
If you have any questions about details within KrabzCAM, please let me know.
Btw. The conversations in the [closed issues](https://github.com/mkrabset/krabzcam/issues?q=is%3Aissue+is%3Aclosed) can also be a good source of information.

Looking forward to more youtube-videos.

davi-domo, 2022-07-15:
> Btw. The conversations in the closed issues [1] can also be a good
> Links:

Hello Marius

I have a small question about the "RELIEF" function

if I understand
"Relief rough" allows you to roughen
"Relief follow" is the finishing pass
I have just discovered the function and suddenly the power of your 
application

I will make the video with chapters for each function
it will therefore allow tutorial and cheat sheet with the chapters

I will make for the tutorial, a pen storage box, with relief engraving 
on the top and certainly laser engraving, which will allow you to use 
all the functions

mkrabset, 2022-07-17:
Hi, yes. The relief-functions are as you describe. 
The "relief rough" allows you to roughen, and also allows stepping down in multiple passes. 
You can set a roughing clearance to leave some vertical clearance for the finish(follow) pass.

The "relief follow" will just follow the z-height given by the image in one pass.

Notice that both functions assume a straight endmill. So if you for instance use a v-bit for the "follow"-pass, it may remove too much material on the sides if the image has 'steep' areas.
Didn't cause too much problems for this example though :-) https://www.flickr.com/gp/25948865@N03/16bk5r7qD0

As for the laser, I'm afraid the "raster" function is not really usable. 
Sorry to disappoint, but this function is just something I put together for my own custom cnc-firmware a while back.
I guess I should hide it, or maybe make it into something similar to lightburn in the future.
The "laser/follow-path" function should be working fine though. You can engrave nice artwork with this as well.

I guess another thing I can recommend trying out is the v-pocket/v-inlay functions. 
I used vpocket with 60deg vbit and some acrylic paint for these: https://www.flickr.com/gp/25948865@N03/Y161oiu560
With the inlay function you can also create wooden inlays, which I think is really nice.
Like these: https://www.flickr.com/gp/25948865@N03/5g7m1A2j82

Good luck with your project!

davi-domo, 2022-07-20:
Hello,
  I will make 2 videos
  the first to describe the functions while making key rings otherwise people will drop out if I make it too complicated
  the second a little later where I would make a pen box allowing to see the power of krabzcam when it is fully exploited and that we have a good visualization of the 3d result while thinking 2d

by realizing this tutorial I discover that I did not even exploit 30% of krabzcam

you should add a link like [buy my cofee](https://www.buymeacoffee.com/) so people can give you thank you donations

davi-domo, 2022-07-29:
hello, 
the video is broadcast from 8 am French time donations from 5 hours from the time of this post
video link: https://youtu.be/IRAATVkPkPY

mkrabset, 2023-01-23:
Hi, I've finally added some documentation.  🎉
Link in the app.

[image]

davi-domo, 2023-01-24:
Hi Marius
Great work very well detailed easy to understand.
I will make a post to subscribers who should appreciate this good news

closed 2023-03-01 by mkrabset