mkrabset/krabzcam #31: [Feature request] - added Roughing clearance function in profile inside and profile outside

davi-domo, 2022-08-02:
hello, 

while making my pencil box for my future video I noticed that the Roughing clearance function was not available in the inside and outside profiles
this function could be useful to create roughing clearances during nesting creation
have a good day

ps: I will post the link of the video once online I finish editing

mkrabset, 2022-08-05:
Hi, great videos!  ( Too bad I didn't learn french at school, but at least I get some some translations in YT subtitles)
I will add a roughing clearance for profiles as well.

davi-domo, 2022-08-05:
Hello
cool thanks for adding the roughing clearance features
for the video I will try to make the subtitles so that the automatic translation is better

mkrabset, 2022-08-05:
I've added the roughing clearance for profile operations.
Please check it out and let me know of any problems.
-M

dadoprom, 2022-08-05:
Hi, I am new to CNC machining... could you please explain to me how to use roughing clearance on for example profile outside... what should it do? Or how should I set the operation... thank you :)

davi-domo, 2022-08-06:
@dadoprom 
hello, 
here are configuration screen prints with and without roughing clearance
this allows for example to make a pass with a straight cutter with a roughing clearance of 3 millimeters then another pass without roughing clearance with a rounded cost to create a rounded bottom of the pocket
[with roughing clearance](https://drive.google.com/file/d/1xXlKtfJdO8-MDFXsdc19mASb0JfMPsFr/view?usp=sharing)
[without roughing clearance](https://drive.google.com/file/d/1-1j8uCxGb3J1C_7cNyHEu834HONFQdFg/view?usp=sharing)

davi-domo, 2022-08-06:
@mkrabset 
hello,
I just did a software test it looks perfect 👍
as soon as I can I do a machine test

davi-domo, 2022-09-19:
hello
I did several tests it works perfectly
on the other hand I can only go up to 4mm in negative
it saves a lot of design time for simple shapes 👍

closed 2022-09-22 by mkrabset