mkrabset/krabzcam #32: Bug in dog-bone feature.

geobruce, 2022-10-04:
I'm trying to use the dog-bone feature and I discovered an issue.

outside profile: (no dog-bones are generated) (project attached)
[profile outside.json.gz](https://github.com/mkrabset/krabzcam/files/9709869/profile.outside.json.gz)
[image]

profile outside inconsistent behavior: [outside inconsistent.json.gz](https://github.com/mkrabset/krabzcam/files/9709926/outside.inconsistent.json.gz)
[image]

Profile inside [dia = 10mm] (no dog-bones are generated) [prof-inside-10mm.json.gz](https://github.com/mkrabset/krabzcam/files/9709927/prof-inside-10mm.json.gz)
[image]

 Profile inside [dia = 1mm] (dog-bones are generated) with the default diameter it works)
[image]

[image]
[dogbone test.zip](https://github.com/mkrabset/krabzcam/files/9709876/dogbone.test.zip)

Can you please have a look at this?
Do you might know a work around or a quick fix?
I'm taking a day off soon to have some time with the CNC machine I'm planning on making this: 
[image]
I included a part of the design file I plan on cutting out. So you can test this. I'm planning on using a 8mm and 3.175mm router bit.
[extensive test file v0.1a.zip](https://github.com/mkrabset/krabzcam/files/9709954/extensive.test.file.v0.1a.zip)

Thanks in advance and keep up the good work!
Please let me know if you get notifications for closed issues. (I'm not sure if this is the right way the re open an issue)

_Originally posted by @geobruce in https://github.com/mkrabset/krabzcam/issues/12#issuecomment-1267477843_

mkrabset, 2022-10-06:
Hi, thanks for reporting!
I've added some fixes. Please do some testing and let me know of any issues.

-Marius

geobruce, 2022-10-11:
Last night I noticed there was an update so I generated the gcode and all seems to work fine for the moment.

Thanks for the quick fix!

closed 2022-11-21 by mkrabset