mkrabset/krabzcam #34: [Feature request] default settings

alanesq, 2023-02-11:
Hi,
I have only just discovered Krabzcam and I must say, it is absolutely brilliant - thanks for creating it.
I built myself a CNC router many years back now and I have been searching for a suitable app to use with it ever since and finally I have found it!

Anyway, what for me would improve it even more would be a way to get it to import my default settings.json file at startup (when running it locally).  I don't believe there is a way to do this at present?
and if it were possible to add my own custom footer and header lines to the gcode it produces

mkrabset, 2023-02-11:
Hi, thanks for your feedback! I've added support for saving default settings.
Two new buttons:
"Set as default" - stores the current settings in browser local storage as new default operations.
"Reset" - loads the 'factory' settings

Please test it and let me know of any problems.

As for your request regarding custom gcode header and footer, I'm afraid you will have to do this by hand for now.
Maybe later.

alanesq, 2023-02-11:
That is excellent, thanks
I just gave it a try and seems to work.  This will make using it much more convenient, especially when in a hurry and forget to load in my settings first...

closed 2023-02-11 by alanesq

dadoprom, 2023-02-11:
wow, amazing, thank you :)