mkrabset/krabzcam #35: Supported Browsers? Not working with older browsers

jeffs555, 2023-04-15:
Krabzcam runs great on my newer PC's, but changes made after Apr 24, 2022 broke krabzcam on older browsers. On older browsers, krabzcam Apr 24,2022 and earlier works fine, but Jun 26, 2022 and later versions just give a blank display. 

Not worth a major effort, but if it is something simple it would be nice to be able to run krabzcam on the same PC controlling my router. 

My router is controlled by LinuxCNC on a laptop, and unfortunately on laptops, LinuxCNC only runs on an obsolete Linux version(Debian 7). The newest browsers I have found for it are from before 2019, Firefox version 63 and Opera version 55. Both browsers work with old versions of krabzcam, but won't work with the current version.

mkrabset, 2023-04-15:
Hi, I've tried to add a few polyfills to support older browsers. Please try the version in branch [issues/35](https://github.com/mkrabset/krabzcam/tree/issues/35).

I don't have any experience with LinuxCNC, but would it be possible to run in in docker perhaps? 
I see there are some docker-images out there..I guess real-time OS requirement may cause issues but maybe still worth checking out...

jeffs555, 2023-04-16:
Thanks for the fast response. That version still fails. Don't know much html and javascript programming, but just ran the debugger in older Firefox and it shows a syntax error. Doesn't flag this as error on new Firefox versions.

with the latest released Krabzcam it gets this error
SyntaxError: identifier starts immediately after numeric literal  -      main.c507ab8c.js:1:1211440

with issues-35 version it gets
SyntaxError: identifier starts immediately after numeric literal -       main.6c32b213.js:1:1213610

also gives this link for more details
https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Errors/Identifier_after_number

As for LinuxCNC, it needs the realtime kernal to drive the stepper motors with the correct timing. For 64-bit capable PC's there is a current 64-bit realtime linux kernal. However, the only laptops with a parallel port to drive the steppers are old and won't do 64-bit mode so I am stuck with obsolete 32-bit linux.

jeffs555, 2023-04-16:
I tried the developer tools in Opera(v55), and it gives the same error as Firefox. With both browsers and both krabzcam versions it points to 0x0n=== as the error. Your code seems to be compiled js so I don't know exactly what that code is doing.

mkrabset, 2023-04-16:
Looks like some of the libraries used relies on [BigInt](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/BigInt) support. Seems at least to be part of the problem. 
(should be supported by Opera >= v54 though...don't know why v55 is not working)

Not an easy fix I'm afraid, so I guess you need to somehow upgrade your browser.

jeffs555, 2023-04-17:
Thanks for that info. It let me know what level of Firefox I needed. Was able to get Firefox v72 installed and Krabzcam is working fine with it. 

That BigInt page shows that there are a couple functions that Opera v54 does not support, and must not be supported in v55. Did download the current Opera version on another PC and Krabzcam works fine with it.

closed 2023-04-17 by jeffs555