mkrabset/krabzcam #36: Dogbones fail on hexagon or higher n-gon.

jeffs555, 2023-04-20:
The dogbone feature is really nice, but it fails if more than 5 inside angles. You can see it by placing an n-gon and doing an inside profile with dogbones enabled. N-gons with 3,4,5 sides work fine and have dogbones in all corners. Hexagon only puts dogbones in 3 corners. If more than 6 sides, then no dogbones are done.

mkrabset, 2023-04-21:
Thanks for your input!
There is a hardcoded maximum angle for dogbones.
I've increased the value from 120 deg to 170 deg. 
Guess I might make this configurable some time in the future.

jeffs555, 2023-04-22:
Thanks, that explains everything I was seeing. I just noticed it because I wanted to do dogbones inside a hex and that change does it. 
Did some playing with triangles and it looks like there is also a minimum angle of around 40 degrees. Don't know if anyone would ever need less than that. 
Thanks again.

closed 2023-04-26 by jeffs555