mkrabset/krabzcam #37: [Feature request] Raster image import resolution

simongcx, 2023-06-03:
Firstly, KrabzCAM is a great tool - thank you.

I've been exploring the laser raster function, and it appears that when importing an image, the software imports at 1 pixel to 1mm, despite offering a much lower "write" resolution (0.25mm appears to be the default). Whilst scaling within the program is possible, that's an extra step in a workflow, and both scaling and positioning are mouse-driven (and hence a little fiddly to get exact). I would typically opt to do any raster image work (compositing multiple images, rotating, scaling, etc) in an image editor before importing to CAM.

I would suggest that having an import resolution parameter would be a useful additional feature for KrabzCAM raster image functionality. This could be implemented as an input box below the "Load PNG / JPG" button, populated with a default value.

mkrabset, 2023-06-03:
Thanks for your input. 
I guess being able to set import resolution would be useful, so I will consider this. 
I'm thinking maybe collect such global settings in a separate dialog instead of on the left toolbar...

However, it is now possible to use keyboard to do precise scaling. 
For instance, in your case if you want to scale the image to 0.25mm/pixel, you can do this:
1) Load the image
2) Type "p" and then "o" to set pivot to origin.
3) Type "s" and then "0.25 <enter>"

Hope this helps a bit :-)

simongcx, 2023-06-03:
Thanks for considering the suggestion.

I'd missed the keyboard controls - they make scaling _much_ easier. It would still be useful to set the import resolution, but the keyboard controls make it much less of a necessity. Thank you!

mkrabset, 2023-10-12:
Hi, I've (finally) added support for setting image import resolution.

New "Settings" button on top, with dialogue:

[image]

closed 2024-07-05 by mkrabset