mkrabset/krabzcam #38: why smaller in KrabzCAM

CycloRetro, 2023-10-04:
Why are SVG files 33% smaller when imported into KrabzCAM?
This bug makes it very difficult to scale things correctly.

davi-domo, 2023-10-04:
Hello
I don’t have that kind of problem
You import can be a svg in inch and krabzcam is in millimeters

mkrabset, 2023-10-04:
Hi, yes, sounds like this is related to metric vs imperial units.
If this is so, I guess the imported shapes should be ~39.37% of the intended size, not 33%. 

I'll look into it and see if there are some improvements to be made.

Should be easy to handle though, just set pivot to origin and scale by factor 2.54.

-Marius

CycloRetro, 2023-10-05:
My KrabzCam settings are set to millimeters but still the exported file is exactly 33.3% smaller. This indicates to me its not a metric vs imperial issue.

mkrabset, 2023-10-05:
Hmm, ok. I guess you need to scale by factor 3 then..
If you post a sample svg, I will have a look.

-Marius

mkrabset, 2023-10-06:
I think I understand now...when you say 33% smaller, I assume you mean the scale is 3/4 of what it is supposed to be. 
So you need to multiply by 4/3, e.g. hit "s" and then "1.3333"+enter. (while mouse pointer is inside the 2d view)

I assume the reason for this is that KrabzCAM (and Inkscape) assume resolution 1inch=96px...while other tools, like Adobe Illustator use 1inch=72px.

I'll put in some words about this in the user manual.

-Marius

CycloRetro, 2023-10-12:
Yes I'm using Adobe Illustrator but in millimetres as its more accurate and easier to work with on small designs. It would be fantastic if SVG files from Illustrator would import into KrabzCAM at the correct size like for like.
Many of my designs are tiny, sometimes there's shapes only 0.01mm and currently KrabzCAM cannot correctly display tiny shapes.

mkrabset, 2023-10-12:
I'm not sure if there is an easy and consistent way to determine original DPI on an svg file.

but..
I've added a new 'Settings'-button.
[image]

...with a dialogue where you can set tolerance (detail level) and svg DPI. 
[image]

The values are stored in the browser local storage, so you only have to set the values once (for each browser).

This is a bit experimental, so don't be surprised if problems occur. :-)
Feedback is appreciated.

-Marius

CycloRetro, 2023-10-30:
Its displaying the tiny shapes well now, thank you. But the sizing issue is still unsolved.

mkrabset, 2023-10-31:
@CycloRetro if you post a sample SVG and a description of what dimension you expect, then I will have a look.

closed 2024-07-05 by mkrabset