mkrabset/krabzcam #40: Starting point outside

Mayov17, 2023-11-23:
Hi, how i can choose the starting point for outside profile milling ? because the starting point actually are not good for me.

Thank you

mkrabset, 2023-11-24:
Thanks for your input!
There is currently no support for choosing starting point of a toolpath following a closed loop.

You might get around it by preparing the svg before loading it though
...like discussed [here](https://graphicdesign.stackexchange.com/questions/102657/set-order-start-of-nodes-in-a-closed-shape-path) with inkscape. 

I guess it's related to this issue: [https://github.com/mkrabset/krabzcam/issues/21](https://github.com/mkrabset/krabzcam/issues/21). 

Would it perhaps help to use attack method _Ramp_ instead of _Plunge_ on your current starting point, so the tool enters the material more gently...if that is the problem ?

-Marius

Mayov17, 2023-11-24:
Hi,

Thank you for have replied speed.
I use plunge and slow speed (80) but the problem are when I have several identical parts machined at the same time the entry point is never in the same place. if I could choose like on cambam I position in a place not visible and always in the same place. I really prefer your software so it would be great to be able to add the option.

Sincerely Krabzcam are the best for me and I really appreciate if this option are available.

Have a nice day and thank you for have replied.

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

mkrabset, 2023-11-25:
Hi, I've added support for changing entry/exit points in profile toolpaths.
Similar to adding tabs, but instead of typing 't', you type **'e'** instead (for entry/exit).

[image]

I hope this is helpful. Please let me know of any issues.

-Marius

Mayov17, 2023-11-25:
Hi,

Thank you so much ! Now Krabzcam are really the best tool for CNC 🙏

Thank you so much for your speed solution

Have a nice day

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

image.png (view on web)<https://github.com/mkrabset/krabzcam/assets/885864/a8682cea-44ea-44c0-9fe2-7e25af971153>

closed 2024-03-13 by mkrabset

davi-domo, 2024-09-28:
> Bonjour, j'ai ajouté la prise en charge de la modification des points d'entrée/sortie dans les parcours d'outils de profil. Similaire à l'ajout d'onglets, mais au lieu de taper « t », vous tapez **« e »** à la place (pour entrée/sortie).
> 
> ![image](https://private-user-images.githubusercontent.com/885864/285575496-a8682cea-44ea-44c0-9fe2-7e25af971153.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjc1NDkzMzksIm5iZiI6MTcyNzU0OTAzOSwicGF0aCI6Ii84ODU4NjQvMjg1NTc1NDk2LWE4NjgyY2VhLTQ0ZWEtNDRjMC05ZmUyLTdlMjVhZjk3MTE1My5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTI4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyOFQxODQzNTlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wNzc1ZWEwYjYwNWVlNmEzMmQ1ODAxNDFhYmY3NGVmOTcxMGZlZGU2ZThhZTNkY2QzMjlmYzFkNWZkOTUxMDk1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.P5NlS8TEsI1RLEo96oQx2KAowc7ghQfUo28G2mlWQyc)
> 
> J'espère que cela vous sera utile. N'hésitez pas à me signaler tout problème.
> 
> -Marius

Hi Marius, a member of the Discord pointed out to me that this function was not in the shortcuts in the background

mkrabset, 2024-09-28:
> Hi Marius, a member of the Discord pointed out to me that this function was not in the shortcuts in the background

Hi David. 
Thanks for the input. I've added it to the background, and to the [user manual](https://mkrabset.github.io/krabzcam/manual/manual.html#_changing_entrypoints).