mkrabset/krabzcam #41: [Feature request] - window close confirmation

davi-domo, 2023-12-16:
hi marius
Is it possible to integrate a confirmation of closing the krabzcam window
so as not to forget to save the current project
good luck and I still love KRABZCAM

mkrabset, 2023-12-16:
Thanks for the suggestion David. 
I've just added a simple solution for this. It can be enabled in the "settings" dialog.
Just check off "Confirm on close" and hit "ok". 
Then it should ask for confirmation before closing or reloading app. 
I've only tested it in firefox and chrome...seems to work fine with those...

[image]

I haven't bothered to check for unsaved changes, so it will ask for confirmation regardless of this..maybe something for later :-)

Please let me know of any issues

dadoprom, 2023-12-16:
Ok, this one was very much needed, thank you :)

davi-domo, 2023-12-16:
Thanks Marius 
It’s really cool and she’s responsive
I’m gonna tell my community
Thanks

closed 2023-12-20 by davi-domo