mkrabset/krabzcam #42: Drag Knife Operation - tool travel might need its own z-height

ibips, 2023-12-18:
Hi Marius, 

I'm trying to use the drag knife operation in Krabzcam. One issue I am having is that when the tool needs to travel to another path, it does so at the default / non-cut z-height (in my case it's 0). For some reason this travel movement still cuts the material. I'm trying to cut a vinyl masking sheet. I understand this could be because of inaccurate setup of the machine ie  bed that is not 100% level, but I'm cutting a 0.2mm thickness material. My setup is a hobby setup, so the bed may not be level to .01 mm accuracy 😵. 

It would be nice if there is a field for the drag knife operation to be able to specify the travel z-height to ensure that during travel the tool/cutter is away from the material. I believe on line 70-71 on the g-code (and is probably another one since it needs to travel to the next path).  I think this is more of a feature request, rather than a bug. Thanks!

[image]
[dragknife.nc.zip](https://github.com/mkrabset/krabzcam/files/13706656/dragknife.nc.zip)
[image]

mkrabset, 2023-12-18:
Hi, thanks for your input. 
I've added a **safetyHeight** (travel height) attribute to the dragknife-operation, with default 1mm above zero.
Hope this is what you need.

Please let me know of any problems.
-Marius

ibips, 2023-12-20:
Wow.. thank you Marius. I've used this new functionality and happy to inform that I have no unintended cut anymore. 
Thank you so much for getting this feature so fast!

closed 2023-12-20 by ibips