mkrabset/krabzcam #43: [Feature request] - tool management

davi-domo, 2024-01-14:
Hi marius,

I wanted to know if you had planned to make a strawberry bank function in order to record the characteristics of the strawberries that you could select after in the course editing and displayed in the comments G-CODE
I know it represents a big development

if there was already a "tool name" field which would be displayed in the G-CODE comment this would allow the opening of the G-CODE in CANDLE or others to easily have the information when opening

Have a good day

mkrabset, 2024-01-21:
Hi, I assume by "strawberry" you mean something like milling tool and not 🍓...  (failed translation of "fraise" perhaps?)  ;-)
..Anyway, I haven't really seen the need for this.

I usually just rename the toolpaths into something descriptive when needed.

Something like this:
[image]

Then you would see these descriptions in the gcode comments..like:

...
...
( BEGIN TOOLPATH 'finish_with_3mm_endmill' )
(   Files:  )
(   Operation:  )
(    name  =  Profile inside )
(    type  =  PROFILE_INSIDE )
...
...

Maybe this could work for you?

I guess a tool description field could be added, but I think I'll hold back for now, 
...trying to keep the user interface as simple as possible.

Best regards
Marius

davi-domo, 2024-01-21:
hi marius

in fact I hadn't thought about it, it's more than enough
Have a good day

closed 2024-01-21 by davi-domo