mkrabset/krabzcam #45: reverse cleanup bug

davi-domo, 2024-03-18:
Hi
I met a gug recently with the cleanups
the pocket is ok but the cleanups are done outside
here are some captures that will be more meaningful
[image]
[image]

mkrabset, 2024-04-09:
Hmm, that's strange. 
Maybe it gets confused about what objects are inside of what objects. 
Seems like one of the object is touching the outer rectangle.
Will it improve if you move the inner parts a tiny bit to the right?

davi-domo, 2024-04-09:
Hi
  I tried with an old project where it worked well, and I found the same problem,
it seems to me that it reverses the path selection

here is the original pocket
[image]

here the cleanup done at the time of the project
[image]

and there the cleanup that I just did again, as we see he did the cleanup on the inside of the lettering
[image]

it's a project that I had saved where everything had worked well
but now it reverses the selections

mkrabset, 2024-04-09:
Thanks for reporting David!
Should be fixed now.

davi-domo, 2024-04-13:
Hello marius
it's perfect I just tried it again
THANKS

closed 2024-04-13 by davi-domo