mkrabset/krabzcam #51: [Feature Request] -  ctrl-z

skoeb, 2024-09-06:
If certain tools are selected (e.g. grab) then `esc` undos the last opperation, however there have been a few occasions where I mistakenly delete a part, or finalize a movement  only to realize I made a mistake. 

This can require a full reload of the svg, which can be quite time consuming. 

The ability to "undo" actions using ctrl-z would be amazing. 

Thank you for this excellent piece of software!

mkrabset, 2024-09-07:
Thanks for your input @skoeb ! 

I've now made some changes to support undo / redo of changes done in the 2d viewport. 
The following operations are supported by the undo history:
- Path selection / deselection
- Path move / scale / normal-scale / rotate    (scaling needs to be invertible, so don't scale any axis to zero)
- Path add / delete 

Changes done to individual path-nodes in edit-mode is not considered.
Operations on images are also not part of the undo history.

I've added a new property "Undo history size" in the Settings, with default 10, e.g. you can go 10 steps back..

There was quite a few changes involved, so I have published it as an experimental version here for now:
https://mkrabset.github.io/krabzcam/krabzcam-experimental/index.html

`ctrl-z` to undo
`ctrl-shift-z` to redo

Please test it, and let me know of any problems

mkrabset, 2024-09-14:
Undo/redo seems to be working as expected. Now included in offical version.

sam-watttime, 2024-09-28:
I had just selected around 100 holes to drill manually, and then fat-fingered "a"... ctrl-z worked like a charm and saved me 10 minutes of lost work. Thanks! Feel free to close this ticket!

closed 2024-10-29 by mkrabset