mkrabset/krabzcam #52: MACH3 support ?

Mayov17, 2024-12-06:
Hello,

I have work during some month with krabzcam for grbl cnc. 
Now i have news cnc with ddcs v4.1 independant controller and this work with mach 3 gcode compatibility. Any chance to choose mach 3 campatibility when i export gcode from krabzcam ?

Because i like krabzcam if i can use it for export gcode compatible with mach3 i really like that !

Thank you so much !

Mayov17, 2024-12-09:
Hello,
i have maybe not explained good but my question are if it's possible to select mach3 post processor ? or possible to add this option. Thank you

mkrabset, 2024-12-09:
Hi @Mayov17 , do you have any info about how MACH3 differs from other controllers like grbl?

I try to keeps things as simple and neutral as possible, but if MACH3 compatibility can be achieved with some simple adjustments or settings, then I might be able to add that.

Mayov17, 2024-12-10:
Mach3 and grbl haven't all same langage for some command. On gcode generator from fusion 360 or other we can select the post processor compatibility (grbl or mach3). I have see it's possible to download post processor on internet. Do you think it's possible to have solution for select the post processor before the generation of the gcode ?

closed 2024-12-19 by Mayov17