mkrabset/krabzcam #53: Choose ball noze

Mayov17, 2024-12-19:
Hello,

It's possible to choose ballnoze in pocket mode ? Because if the program haven't the information of i use ball noze i haven't full pocket (some material are not removed on my pocket).

Thank you

mkrabset, 2024-12-21:
Hi, no there are currently no option for ballnose, and pocket-operations are primarily intended for flat endmills.

But you have a "stepover" parameter (measured in % of tool diameter) on pocket operation, so you should be able to get your desired result with some calculation.

You can find both tables and online calculators for this online. 
Just search for "scallop height calculator".

Mayov17, 2024-12-21:
Hello,

Thank you for your reply. I have test with stepover option but no good résultat. I need more tests.

If one day an option are add for choose the radius of the tool that are amazing.

Thank you

Mayov17, 2025-01-06:
After verification it's Bullnose (0.5 and 1 radius) i need. I have made some test with step over option but i lost important number of materal before find good solution with stepover. No people can creat code for add Bullnose (radius) option ? that need big modification on the code or i can find one people for make that ?

Thank you

mkrabset, 2025-01-07:
Ah, bullnose not ballnose. That's a bit different.
If I have understood correctly, the flat part of a bullnose is 

`d-2*r` , where `d` is tool diameter and `r` is corner radii.

So if you calculate `(d-2r)/d`, you get a factor you can use to get a stepover%-value adjusted for your tool.

Example:
`d = 6mm`,   6mm tool diameter
`r = 1mm`,    1mm corner radius

flat part = 6mm-2*1mm = `4mm`
factor = 4mm/6mm = `0.66667`

Desired stepOver=40%  (default, but meant for flat endmill)
Adjusted stepOver=40% * 0.66667 = `26.667%`

This is something I leave out of the KrabzCAM application, at least for now. 
The reason is I try to keep the user interface from becoming too complex.